meshtastic/web #283: [Feature Request]: Show two-way traceroute with SNR per hop if available

GUVWAF, 2024-08-29:
### Description

Since [2.5](https://github.com/meshtastic/protobufs/pull/563), when performing a traceroute, nodes will record the SNR per hop, as well as the route back. It would be nice to show this if available. For example, it could show:
```
1 Hop towards, 2 Hops back
Route towards: You --> Node 1 (-10.75dB) --> Node 2 (-5.25dB)
Route back:    Node 2  --> Node 3 (-2dB) --> Node 1 (-1.5dB) --> You (-8dB)
```

Since old firmware won’t add the SNR and the route back, there is some validation needed. For the SNR to be valid, the SNR list should be one longer than the route list (because the destination adds SNR as well). For the route back to be valid, also `hopStart` needs to be populated, because only then nodes can add unknown nodes and SNR. Unknown SNR is represented by the value -128 and unknown nodes by `0xFFFFFFFF` (4294967295). The SNR values are in dB, but they are scaled by 4, so it has to be divided by 4 before showing it. A reference implementation in Python can be found [here](https://github.com/meshtastic/python/blob/2.5/meshtastic/mesh_interface.py#L549).

philon-, 2025-05-11:
Correct me if I'm wrong, but I believe this has been added since the creation of this issue.

This is the current traceroute response dialog:

[image]

GUVWAF, 2025-05-11:
Yes, indeed! Thanks!

closed 2025-05-11 by GUVWAF