meshtastic/web #305: [Bug]: Multiselect for Position Flags is broken

Hunter275, 2024-09-17:
### Hardware

Not Applicable

### Connection Type

Serial

### Local or Hosted

http://meshtastic.local, https://client.meshtastic.org

### Firmware Version

2.5

### Description

[image]

This is supposed to be a multiSelect

https://github.com/meshtastic/web/blob/4d1227a9427fb1216f11737548daf380701226a7/src/components/PageComponents/Config/Position.tsx#L53-L60

Jason-Patrick, 2024-09-26:
Just for additional information: the multiselect functionality is missing here
https://github.com/meshtastic/web/blob/06d2c393ce4d83bfb42158e333dbd5362a0c0c34/src/components/Form/DynamicFormField.tsx#L60-L61

danditomaso, 2024-12-30:
Wanted to see if a multi-select is the best control to use for this field? Would a dropdown with checkboxes also be sufficient? There is a Shadcn demo version in this Vercel V0 sandbox https://v0.dev/chat/PCSLWjh2drb.

PR #394 (2025-02-08): Feat: Add MultiSelect component
- Created reusable MultiSelect checkbox component
[image]
[image]

- Created custom hook to manage position flag state (usePositionFlags)
- Internal flag state management with computed activeFlags
- Type-safe flag manipulation (toggleFlag, setFlag, setFlags)

Fixes #305
closed 2025-02-08 by Hunter275