meshtastic/web #316: [Bug]: Publicly hosted web client fails to connect and fetch device information

hammyMarc, 2024-10-12:
### Hardware

T-Beam

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.5.5.e182ae7 Stable Beta

### Description

I'm a hacker with enough programming experience (professionally) to be dangerous. So with that said....
Windows 10 - tested Brave and Edge with the same failed result. I can pring the t-beam device. I can connect to it over serial. I can connect to it via the Android app via IP. But I can't seem to connect to the mesh device using https via a browser.

At least the failure is easily repeatable on different platforms and browsers (Linux Mint and Win10 with Brave and Edge). See image which pinpoints the line of code at which it consistently fails.

[image]

Talie5in, 2024-10-12:
Connect to your device first (https://meshtastic.local) and accept the self signed cert, then try via client.meshtastiic.org

closed 2024-10-12 by sachaw

hammyMarc, 2024-10-12:
Talie5in:

I can’t seem to build a proper SSL connection to the device. I did add the device cert to Windows. No help.

***@***.***

***@***.***:~# openssl s_client -connect meshtastic.local:443
40E7D0D57D7D0000:error:10080002:BIO routines:BIO_lookup_ex:system lib:../crypto/bio/bio_addr.c:738:Name or service not known
connect:errno=2
***@***.***:~# openssl s_client -state -nbio -connect meshtastic.local:443 2>&1 | grep "^SSL"
***@***.***:~# openssl s_client -state -nbio -connect meshtastic.local:443
40A7FF15EF7E0000:error:10080002:BIO routines:BIO_lookup_ex:system lib:../crypto/bio/bio_addr.c:738:Name or service not known
connect:errno=2
***@***.***:~# openssl s_client -state -nbio -connect www.google.com:443<http://www.google.com:443>
CONNECTED(00000003)
Turned on non blocking io
SSL_connect:before SSL initialization
SSL_connect:SSLv3/TLS write client hello
SSL_connect:error in SSLv3/TLS write client hello
write R BLOCK
SSL_connect:SSLv3/TLS write client hello
SSL_connect:SSLv3/TLS read server hello
SSL_connect:TLSv1.3 read encrypted extensions
depth=2 C = US, O = Google Trust Services LLC, CN = GTS Root R1
verify return:1
depth=1 C = US, O = Google Trust Services, CN = WR2
verify return:1
depth=0 CN = www.google.com<http://www.google.com>
verify return:1
SSL_connect:SSLv3/TLS read server certificate
SSL_connect:TLSv1.3 read server certificate verify
SSL_connect:SSLv3/TLS read finished
SSL_connect:SSLv3/TLS write change cipher spec
SSL_connect:SSLv3/TLS write finished
read R BLOCK
---
Certificate chain
0 s:CN = www.google.com<http://www.google.com>
   i:C = US, O = Google Trust Services, CN = WR2
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: RSA-SHA256
   v:NotBefore: Sep 24 03:18:37 2024 GMT; NotAfter: Dec 17 03:18:36 2024 GMT
1 s:C = US, O = Google Trust Services, CN = WR2
   i:C = US, O = Google Trust Services LLC, CN = GTS Root R1
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Dec 13 09:00:00 2023 GMT; NotAfter: Feb 20 14:00:00 2029 GMT
2 s:C = US, O = Google Trust Services LLC, CN = GTS Root R1
   i:C = BE, O = GlobalSign nv-sa, OU = Root CA, CN = GlobalSign Root CA
   a:PKEY: rsaEncryption, 4096 (bit); sigalg: RSA-SHA256
   v:NotBefore: Jun 19 00:00:42 2020 GMT; NotAfter: Jan 28 00:00:42 2028 GMT
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIEWDCCA0CgAwIBAgIQMsnnZM7+yJIQ/4sBn5GTdzANBgkqhkiG9w0BAQsFADA7
MQswCQYDVQQGEwJVUzEeMBwGA1UEChMVR29vZ2xlIFRydXN0IFNlcnZpY2VzMQww
CgYDVQQDEwNXUjIwHhcNMjQwOTI0MDMxODM3WhcNMjQxMjE3MDMxODM2WjAZMRcw
FQYDVQQDEw53d3cuZ29vZ2xlLmNvbTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IA
BIYyYxNBimBZKERlKxeJWa+0T9ChGca0HOk55zc6VQr7y9u6jxIxpUxkoli3qvmG
an3KMnUlObbAB+cVNlOnMZWjggJDMIICPzAOBgNVHQ8BAf8EBAMCB4AwEwYDVR0l
BAwwCgYIKwYBBQUHAwEwDAYDVR0TAQH/BAIwADAdBgNVHQ4EFgQUeDNA9UA8F4Pb
NW2o5iOLegKWQAIwHwYDVR0jBBgwFoAU3hse7XkV1D43JMMhu+w0OW1CsjAwWAYI
KwYBBQUHAQEETDBKMCEGCCsGAQUFBzABhhVodHRwOi8vby5wa2kuZ29vZy93cjIw
JQYIKwYBBQUHMAKGGWh0dHA6Ly9pLnBraS5nb29nL3dyMi5jcnQwGQYDVR0RBBIw
EIIOd3d3Lmdvb2dsZS5jb20wEwYDVR0gBAwwCjAIBgZngQwBAgEwNgYDVR0fBC8w
LTAroCmgJ4YlaHR0cDovL2MucGtpLmdvb2cvd3IyLzc1cjRaeUEzdkEwLmNybDCC
AQYGCisGAQQB1nkCBAIEgfcEgfQA8gB3AO7N0GTV2xrOxVy3nbTNE6Iyh0Z8vOze
w1FIWUZxH7WbAAABkiJAtHYAAAQDAEgwRgIhAPKCWg/ABhXkKrCrTEP2ihohiKzP
LWcFjnDnQ+sR8yfOAiEAmmzy3h+Umhk/Z0Jn2Q0tdjMkkFcXlMhB8Wflvse6nuIA
dwBIsONr2qZHNA/lagL6nTDrHFIBy1bdLIHZu7+rOdiEcwAAAZIiQLUMAAAEAwBI
MEYCIQCGjGuIgmPe7Q+CgobC9czqLt5z5ql3cQafOJTaLaRuDgIhAKcQiBCb6OmA
ZEvmwBgIjRqJ6M8tgzVWw4ooHrjBLVWUMA0GCSqGSIb3DQEBCwUAA4IBAQClRLQM
rqYSc7jo5CKuVWzYTzemRQ4equj8f2NzRGwQGqsXnbNalM8qVrNIj6jXPIFMjB2s
+zx1olIywJj3sHCHs/7e3PFmzcdiYldIWJf0vOsxAugudjRApUDRXV7lbo8gqQ57
FmDbc9g8x9iCAnBEmodmvLo/l+sYG3LP05tVS2eZOguhjv/KYqo2+OE0WqrBl+nA
1u4tytjvrs2dO4kzQGtEX2PGpCjfugmg6J30pGq9oTVu9QQTATcqZfF1ze6AgxOG
dDTKBJQN/z/A6Cglv3C6aqmMB/Ngl7Ub6IiBxQgD48qqALN65ewkGT82KosX3n5t
g54nGAwPv2m8YZ+Q
-----END CERTIFICATE-----
subject=CN = www.google.com<http://www.google.com>
issuer=C = US, O = Google Trust Services, CN = WR2
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 4108 bytes and written 396 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
SSL3 alert write:fatal:decode error
40C7701ED67D0000:error:0A000126:SSL routines:ssl3_read_n:unexpected eof while reading:../ssl/record/rec_layer_s3.c:317:
***@***.***:~#

hammyMarc, 2024-10-12:
Not sure if this is helpful @Talie5in  but here is a capture via Wireshark.

[image]

hammyMarc, 2024-10-12:
It would appear that this is an issue on the device itself. @sachaw do you feel this is or is not an issue in the code?

More info:

[image]

SSL_new failed. Aborting handshake. FID=52

sachaw, 2024-10-12:
If you visit https://device_ip where device_ip is that of your device. Do you get a self signed TLS cert warning?

hammyMarc, 2024-10-12:
Yes, I did.

  1.  I selected “Advanced” then “Proceed”. The connection would then get reset by the t-beam (multiple times). Eventually, the browser gives up.
  2.  I right-clicked on the “security icon” left of the Address Bar and downloaded the public .crt. I added that cert to the Windows Cert Storage. Restarted the browser, attempted a connection to the HTTPS server and again the t-beam device reset / closed the connection attempt.

If you think it’s a hardware issue, I will return the device.

Thanks,

Marc

hammyMarc, 2024-10-14:
I'm convinced that this is a real bug in software. My evidence for such is:
1) I can access the device over HTTP but not HTTPS / TLS.
2) I've provided proof Wireshark captures and Dev Tools that the device is just slamming shut the TCPIP connections (repeatedly) in spite of the fact that I select "Advanced" and "Proceed" in a Chrome broswer. It's not a client-side (browser or firewall or network) issue.

Hope that helped. I would encourage this issue to be investigated and resolved, if the group wants to support TLS between the mesh device and various IP network clients.

Cheers. This is an interesting project.

Talie5in, 2024-10-16:
@hammyMarc Can you do a full erase and flash, going to https://local_ip should load a local version of the webclient.

Just tested this on a tbeam supreme and a t3s3 on 2.5.4 and this works as expected.

hammyMarc, 2024-10-16:
@Talie5in I am grateful for your curiosity and support in this matter.

I did a full erase and flash of the t-beam to 2.5.4, as requested. Then:
1) Paired my phone via BL.
2) added my wifi parameters.
3) attempted to connect via https://my_ip.

The same error has been produced. 

[image]
[image]

[image]
[image]
[image]

[image]
[image]

I did some looking around and found this article. I hope is it helpful:

https://github.com/fhessel/esp32_https_server/issues/140

That issue appears to mirror our issues, verbatim.