meshtastic/web #318: [Bug]: Leading zero removed from entered Bluetoth PIN

zl3ag, 2024-10-15:
### Hardware

T-Lora v2 1.6

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

2.5.5

### Description

Head to the Bluetooth config screen, type in a 6 digit number that starts with 0. After saving the leading 0 has gone and you're left with a 5 digit PIN.

garthvh, 2024-10-16:
The BLE pin has to be six digits and can't start with a 0

zl3ag, 2024-10-17:
So why does it let you enter a number starting with 0?

garthvh, 2024-10-17:
It shouldn't

luf7, 2024-10-31:
I don't think this is a bug in the web client per se, but by design to work with the meshtastic firmware. Seems like all you can do is live with it and use a PIN without a leading zero. An info text next to the field that informs about the format requirement or a bit more validating would be nice though, for example I've tried setting the PIN to 4294967295, the biggest 32-bit integer value and it made my client freeze, I guess memory leak.

PR #338 (2024-11-10): Add bluetooth pin validation
Fix #318
closed 2024-11-10 by fifieldt