meshtastic/web #328: [Bug]:

andy1v, 2024-10-27:
### Category

BLE

### Hardware

T-Deck

### Firmware Version

2.5.3

### Description

In web client can not Enable Bluetooth
After you enter in settings  bluetooth and enable, press button save device will restart in next 20 sec and bluetooth is still disable

caveman99, 2024-10-28:
If Wifi is enabled bluetooth will stay disabled. They can not be on at the same time. Limitation of the Hardware.

andy1v, 2024-10-31:
I will double check,but just like I remember wifi was off
Thank you

andy1v, 2024-11-01:
Team,
I double check and wifi is off and still [is the same issue 
When enable bluetooth and save config device is going to reboot an is not activating bluetooth 

Thank you

Hunter275, 2024-11-10:
Bug confirmed

Hunter275, 2024-11-10:
@andy1v 

I cannot replicate this issue, please confirm that WiFi is DISABLED like in the screenshot:
[image]

Save the settings and let the device reboot THEN enable Bluetooth and save the settings.

closed 2024-12-30 by Hunter275

tomhanax, 2025-02-04:
Same problem here on 2.5.21. Connecting via serial using client.meshtastic.org. Wifi off, ble off. When I enable ble an save, device restarts but still no ble and gui switcher is off.