meshtastic/web #333: [Bug]: Web Interface Missing

darkdragone, 2024-11-04:
### Hardware

Heltec V3

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

2.5.10.0fc5c9b

### Description

Entered IP address and meshtastic.local into chrome both with https and without. I get a page that says "Web server is running.

The content you are looking for can't be found. Please see: FAQ.

admin"

I've tried a full erase using the web flasher originally with version 2.5.9.936260f and with 2.5.10.0fc5c9b

matthewCmatt, 2024-11-05:
Thanks for filing. Seems some users on Discord reported this on firmware:
2.5.5
2.4.2.5b45303

One user reported rolling back from 2.5.5 to 2.5.4 to fix

darkdragone, 2024-11-08:
> Thanks for filing. Seems some users on Discord reported this on firmware: 2.5.5 2.4.2.5b45303
> 
> One user reported rolling back from 2.5.5 to 2.5.4 to fix

Haven't tried that but earlier versions had a battery bug for the V3. Hopefully the web interface will be fixed in future versions.

closed 2024-11-08 by thebentern

Sjoerdfc, 2025-03-09:
I'm still having this issue. I flashed my Heltec Wireless Tracker with the latest stable, 2.5.20, and the only thing I saw on the web interface was the "Web server is running" message. I'm flashing with the device-install.sh script in Linux.

I then flashed 2.5.23, same issue, and then also with 2.5.18 the same issue. I did not update and did a clean device-install everytime.

I  then reverted to 2.5.4, and finally I have a web interface.

fifieldt, 2025-03-10:
In the flasher you need to hit "full erase" and then "include webserver"

Sjoerdfc, 2025-03-10:
I'm using the shells script provided with the firmware files to flash. I just noticed there is a --web option:

```
./device-install.sh --help
Usage: device-install.sh [-h] [-p ESPTOOL_PORT] [-P PYTHON] [-f FILENAME|FILENAME] [--web]
Flash image file to device, but first erasing and writing system information"

    -h               Display this help and exit
    -p ESPTOOL_PORT  Set the environment variable for ESPTOOL_PORT.  If not set, ESPTOOL iterates all ports (Dangerous).
    -P PYTHON        Specify alternate python interpreter to use to invoke esptool. (Default: "/usr/bin/python3")
    -f FILENAME      The .bin file to flash.  Custom to your device type and region.
    --web            Flash WEB APP.
```
I think that's a new option then, it wasn't there in earlier versions I think. This isn't in the documentation on the site though:
https://meshtastic.org/docs/getting-started/flashing-firmware/esp32/cli-script/

Maybe --web should be added to the command examples on the site then? Because the next page assumes the web interface was installed.