meshtastic/web #345: [Feature Request]: Specify Host in URL

Yeraze, 2024-11-12:
### Description

Can we get some mechanism of specifying the host in the URL?  Even better, multiple hosts?

Right now, in order to load my 3 nodes in the Web UI (all reachable via static IP's over wifi) I have to spend a good 15-30s clicking around all the Add/Enter URL/Connect, etc.  Would be great to just bookmark the fully populated UI and let it start the retrieval.  

Even just a simple "?local" to default the connection to the same host as the webserver would be a huge benefit.

kquinsland, 2025-01-25:
+1 for this as well. 

Ideally, the SPA would use something like [`window.localStorage()`](https://developer.mozilla.org/en-US/docs/Web/API/Window/localStorage) to store the connection(s) in a manner similar to cookies. This way I wouldn't have to re-type the 3 hostnames... _every time_ I load up the web client :/

danditomaso, 2025-05-04:
@Yeraze @kquinsland So I'm looking into adding this feature into the next release. This would only make sense to use with HTTP connection method. Do you feel it should also support bluetooth connections as well?

kquinsland, 2025-05-04:
> This would only make sense to use with HTTP connection method. 

I don't follow? Does the web bt api not give you the unique mac address or whatever needed to uniquely identify that connection in the future?

> Do you feel it should also support bluetooth connections as well?

Yes. I haven't figured out precisely where the bug(s) are, but i often find that some/all of the changes made via the web ui do not persist when saving. I have gotten into the habit of making _precisely exactly one change_ and then saving it and then `cmd + r` to refresh and re-connect to the device (either bt or usb/serial) and to verify that the change was actually made. Annoying, but would be considerably less annoying if I didn't have to go through the "new device, do you want chrome to connect to? ..." flow every time.

atrent, 2025-08-31:
I have three nodes (reachable by wireguard), it would be super useful to bookmark them!