meshtastic/web #348: [Bug]: Missing Attribution for Openstreetmap Tiles

5aji, 2024-11-23:
### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local, https://client.meshtastic.org

### Firmware Version

a147324

### Description

According to the [Tile Usage Policy](https://operations.osmfoundation.org/policies/tiles/) of the `tile.openstreetmap.org` server:
>  - Clearly display [license](https://wiki.openstreetmap.org/wiki/License) [attribution](https://wiki.osmfoundation.org/wiki/Licence/Attribution_Guidelines), normally in the bottom-right corner of the map.

At present, there is no attribution visible on the web map. I believe MapGL has an option to enable this, so it should be fairly straightforward.

Furthermore, there are other recommendations on the Usage Policy page that may apply and can help:
> - Recommended: Do not hardcode any URL to tile.openstreetmap.org as doing so will limit your ability to react if the service is disrupted or blocked. In particular, switching should be possible without requiring a software update.
> - Recommended: add a link to https://www.openstreetmap.org/fixthemap to allow your users to report and fix problems in our data. 

The hardcoded URL is handled with #33, but perhaps the attribution in the corner of the map should also have a link to make edits.

danditomaso, 2024-12-28:
I can take this on if no one else is working on it. I have experience working with the React Map GL library.

fifieldt, 2024-12-28:
That would be great!

danditomaso, 2024-12-29:
Happy to help out! The PR is ready and is waiting for approval #356

danditomaso, 2024-12-30:
This item should be closed as PR #356 was merged.

closed 2024-12-30 by fifieldt