meshtastic/web #352: [Bug]: self-hosted web Bluetooth not enabled

auggie-lahey, 2024-12-04:
### Hardware

Other

### Connection Type

Bluetooth

### Local or Hosted

http://meshtastic.local

### Firmware Version

NA

### Description

spin up docker container
go to http://0.0.0.0:8080/
connect new device
bluetooth connection is greyed out and unable to be clicked

### Relevant console output

```Shell
i have connected and confirmed this works in this browser with client.meshtastic.org
```

Lebbitheplow, 2024-12-16:
Also having this issue, Getting a message "Web Bluetooth and Web Serial are currently only supported by Chromium-based browsers." while on chrome.

ianmcorvidae, 2024-12-16:
The error message possibly does not say it, but it's a browser restriction that web serial and web bluetooth are supported only on https connections.

If you set up SSL and such through a reverse proxy with a certificate, you should be able to use it with a self-hosted copy, though that can be somewhat of a pain depending how you're doing things.

Lebbitheplow, 2024-12-16:
You're definitely right about that one! Configured it under my reverse proxy and it works, Thank you.

danditomaso, 2024-12-30:
Just following up on this item, does this mean the current issue can be closed if a new issue was opening requesting the error messages be improved to avoid situations like the one above?

Hunter275, 2024-12-30:
> Just following up on this item, does this mean the current issue can be closed if a new issue was opening requesting the error messages be improved to avoid situations like the one above?

We'll use this issue.

Hunter275, 2024-12-30:
@ianmcorvidae Can you confirm the new reading of the issue please?

Hunter275, 2024-12-30:
@danditomaso Are you interested in doing this enhancement?

danditomaso, 2024-12-30:
Yeap, it's something I can tackle after the current item I'm working on.

ianmcorvidae, 2024-12-30:
If you mean the updated title, sounds about right. The restriction doesn't just apply to self-hosted instances, I suppose, but I think the hosted one is already https only so it really only comes up with self-hosting

Hunter275, 2024-12-31:
> If you mean the updated title, sounds about right. The restriction doesn't just apply to self-hosted instances, I suppose, but I think the hosted one is already https only so it really only comes up with self-hosting

Good catch, I've updated the title and such.

danditomaso, 2025-01-05:
Just wanted to double check on this item, I did read that the web serial API does require HTTPS, but is there an exception when connecting on localhost? Using a local copy of the Meshtastic web app, I haven't encountered issues connecting  using Web Bluetooth or Serial to my Heltec V3 while on localhost

danditomaso, 2025-01-06:
The PR for this issue is ready to go

danditomaso, 2025-01-07:
@Hunter275 Wanted to see if you'd have a chance to review this PR this week?

PR #362 (2025-01-20): fix: update connect dialog messaging to describe requirement for https
- Consolidated browser feature detection into typed hook

Fixes #352 

[image]
[image]
closed 2025-01-20 by Hunter275