meshtastic/web #361: [Bug]: HTTP API DoS / No back-off on API failures

broglep, 2025-01-05:
### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

-

### Description

The web interface constantly performs new HTTP API calls even if they have failed. There seems to be no time in between requests and this essential performs an denial of service to the API and at the same time drives up CPU usage of the browser.

Web web interface should add some delay after failed request, ideally an exponential backoff and maybe even treat 4xx and 5xx differently. For 4xx response codes it is unlikely that retrying will help anything, so web interface should probably abort. For 5xx response codes the API call can be retried after some backoff

danditomaso, 2025-01-05:
So I had a quick look at the data fetching within the web client and found its handled by the Meshtastic JS library, and I'm not trying to pass the buck here or anything, but I know it would be resolved faster if this issue  was created in that repository. 

https://github.com/meshtastic/js/

danditomaso, 2025-01-05:
Thanks for the report! Can you please open this issue in the JS library repo : https://github.com/meshtastic/js ?