meshtastic/web #368: [Bug]: Running self-hosted container on Raspberry Pi fails

Dezorian, 2025-01-13:
### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

not applicable

### Description

I'm trying to host the container using podman on my Raspberry Pi CM4. I've got podman running and working with the httpd and standard nginx container (example for nginx). But the docker from Meshtastic does not run at all. No logs, nothing. I've run --log-level=debug, and that just looks fine, but for some reason the image just does not run. So I naturally get connection-reset replies when using curl.

```
~ $ podman run -dt --name nginx -p 8080:80/tcp docker.io/library/nginx
de9af0ce6bcf74b151ec90359cab663e7d26d4aed1edf2013a1ae0991f15a407

~ $ curl 127.0.0.1:8080
<!DOCTYPE html>
<html>
<head>
<title>Welcome to nginx!</title>
<style>
[...]

~ $ podman run -dt --name Meshtastic-Web -p 8888:80/tcp ghcr.io/meshtastic/web
bc0a7ba476c8dc517ebebbcd8308aa7f6a012155291c9591c794e0cfd3d48bb5

~ $ curl 127.0.0.1:8888
curl: (56) Recv failure: Connection reset by peer

```

tdhawk, 2025-01-18:
I just ran into this as well. The container is actually listening on port 8080. Change your port mapping to `8888:8080`

I just submitted a pr to update the readme.

danditomaso, 2025-01-24:
@Dezorian It sounds like your issue is now solved? Can we go ahead and close this issue if that is the case?

closed 2025-01-24 by Dezorian