meshtastic/web #371: [Bug]:

DrBlackross, 2025-01-13:
### Hardware

Heltec v2.1

### Connection Type

Serial

### Local or Hosted

http://meshtastic.local, https://client.meshtastic.org

### Firmware Version

lastest?

### Description

install docker meshtastic on rpi, goto serial on the meshtastic client webpage (hosted in the docker container), the serial port is not found.... even after passing docker the serial port(s) where the lora is located locally in the system.

the docker image has no lsusb or ability to update the docker image at all via remote root console, soo cant install lsusb or update the docker image internally.

kinda stuck trying to make it a stand alone in the corner mesh box, that i can add to the local net and check via hostname.local.

### Relevant console output

```Shell
none really...
```

caveman99, 2025-01-13:
Please do not open issues more than once. This is already in the python repository. https://github.com/meshtastic/python/issues/729 - it s also not an issue with the web client but with the way docker handles serial ports. i don't see no easy solution to this sorry. this is a shortcoming of docker, you'll find out if you install on bare metal, it works.

closed 2025-01-13 by caveman99

DrBlackross, 2025-01-15:
k