meshtastic/web #373: [Feature Request]: Web docker local node access

robot-stefan, 2025-01-18:
### Description

Not sure if this is possible and I am simply not seeing it in the documentation anywhere or if its not yet supported. 

I have a computer running portainer and docker. I have tried to pass over both /dev/ttyACM0 and /dev/bus/usb/###/### I can't seem to get it to list the meshtastic node the same way it would be listed on a webui running on a node. 
1. Are there additional environment variables or a 
2. specific way I need to pass through the USB or serial ie to /dev/something_else or 
3. Is this not presently supported function?

Will have this connected over USB for power and doing this would in theory allow for connection over bluetooth with the app from a mobile device. The node I am using does not allow for bluetooth and wifi connection at the same time. 

Chart below for added clarity on setup. 

```mermaid
  graph TD;
      A:[PC]-->|Network - wifi or ethernet|B:[Portainer Server with docker container];
      B:-->|USB or Serial|C:{Meshtastic Node};
```

Thanks,

ianmcorvidae, 2025-01-18:
to support a case like this, you would need to proxy the client API from the serial connection to an HTTP connection; the web client doesn't have a server-side component at present that does this. I have a very rough proof of concept of this sort of proxy at https://github.com/ianmcorvidae/meshtastic-http-proxy but it would need work for a high quality integration for sure. if you're willing to deal with the here-be-dragons nature of it, perhaps it can meet your use case, though!

robot-stefan, 2025-01-19:
Thanks for the clarification on this functionality. I'll try to look more at this later. Not sure I'd be getting it up and running. Looks like I would need to add something for USB/serial connection to the linked docker container? For the time it does seem simpler to experiment via bluetooth / serial.

danditomaso, 2025-04-05:
@robot-stefan @ianmcorvidae Please see this discord screenshot regarding if this is possible. I'd recommend you check out the meshtastic-http-proxy project linked below and comment in that repo, as the requested functionality wont be offered in the web client. https://github.com/ianmcorvidae/meshtastic-http-proxy

[image]

robot-stefan, 2025-04-05:
I'll look at this again. The readme didn't make it quickly clear what it is doing / the objective and how it fit with the docker image here.