meshtastic/web #386: Can't connect via IP address

Sawadee2u, 2025-01-28:
Having the same issue (Can't connect to unit {LiliyGo T-Beam Supreme V3.0) using IP address. Was working when first programmed but after running for three days and trying to connect again using IP address, can't connect. Have IP address being assigned by router and see it on the last screen of device. Also connected to on-line map using MQTT (Green dot on https://meshtastic.liamcottle.net/). Have tried Win11 edge and Chrome. Have reprogrammed with and without erase with Mashtastic flasher without improvement. Using Mashtastic Ver.2.5.18. Not sure what else to try.

Sabering1, 2025-01-29:
I had a same issue, if u flashed new firmware without bundling the webui, you wont be able to use the hosted version of meshtastic client, as it cannot generate a cert for you, however u can run and connect if you selfhost it yourself

Sawadee2u, 2025-01-29:
Thanks for the heads up. I have never checked that box because of the caution about memory use. So am confused it worked when I first programmed. But re-flashed with this checked per your help and all works.Thanks* * * * * * * * * * * * * * * * * * * * * * * * *We all say we do the best we can.We really do the best we want to.* * * * * * * * * * * * * * * * * * * * * * *

closed 2025-01-30 by fifieldt