meshtastic/web #389: [Bug]: Admin Key not taking in Web UI

TheAZtech, 2025-01-29:
### Hardware

Not Applicable, Other

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.5.20

### Description

client.meshtastic.org > Config > Security > Admin Key. 

When you put a key in this box and click save, refresh and nav to the same location, it remains blank. I am having to use the iOS app to input the key and then returning to this screen the key is present. 

This is occurring across several types of devices and firmware versions. 

[image]

### Relevant console output

```Shell

```

danditomaso, 2025-02-02:
@TheAZtech Can I ask when you enter a key in this box, is the Managed toggle enabled? Just asking because we are looking to fix this entire section and ensure users understand what they are toggling.

TheAZtech, 2025-02-02:
No, the Managed Toggle is not enabled. I learned my lesson with that one and don't touch that toggle any longer. 😅

danditomaso, 2025-02-02:
@TheAZtech Fair enough, i've been there and done that 👍 . I might be wrong on this, but the Admin Key field needs to be used in combination with the Managed toggle, since the toggle controls if a specific node is controlled by a central management node. 

I guess I should ask, without the node being set to "Managed" what were you trying to achive using the Admin key field?

TheAZtech, 2025-02-04:
And that makes perfect sense, I am just not as smart as some with CLI. 😅

I personally use the web ui to initially program all of my nodes, then move to the apps. I like the layout better then in the apps. 

The main node I use admin stuff on is on a tower on a local mountain, which then can be controlled via CLI from my base that is LOS. My understanding of CLI is limited and I use it to pull basic information from the node, not make major changes. 

If there is ever a reason I have to run a usb cable up the tower to make changes to the programing for something that CLI can not do, I would not have to figure out how to disable the that admin block to be able to do so.  

I hope that makes sense.

PR #460 (2025-02-26): fix: admin key can be saved/restored inside of web ui.
This PR refactors the Security component to improve state management and validation handling, including fixing a known issue with the admin key field. 

The changes include:
- Replacing multiple useState hooks with a single useReducer for centralized state management.
- Used error management from appStore hook for more consistent and standardized error handling across the application
- Allow for direct key generation for the admin key field. 
- Added elements from the private key password generator to the admin key field to allow for impro [... cut]
closed 2025-02-26 by danditomaso