meshtastic/web #402: [Bug]: User is able to type messages longer than limit

Hunter275, 2025-02-04:
### Hardware

Not Applicable

### Connection Type

HTTP, Bluetooth, Serial

### Local or Hosted

http://meshtastic.local, https://client.meshtastic.org

### Firmware Version

2.5.X

### Description

Its currently possible to type a message that is longer than the Node can send, this results in the message appearing to send just fine but the Node never sends the message.

### Relevant console output

```Shell

```

danditomaso, 2025-02-04:
What is the current byte limit of a message?

garthvh, 2025-02-04:
Clients use 200 bytes

danditomaso, 2025-02-04:
Should be easy enough to limit/prevent a user from entering more than characters

Hunter275, 2025-02-04:
@danditomaso I was planning on tackling this tonight.

danditomaso, 2025-02-04:
> @danditomaso I was planning on tackling this tonight.

Sounds good.

Hunter275, 2025-02-05:
@garthvh 

Is the 200 byte limit defined in the protobufs anywhere?

I found [this DATA_PAYLOAD_LEN](https://github.com/meshtastic/protobufs/blob/b80785b16bc0d243b97917998706e7bf209cd9d0/meshtastic/mesh.proto#L1304), but its not accounting for entire payload.

closed 2025-02-06 by fifieldt

garthvh, 2025-02-06:
It is not in the protobufs, DMs and broadcast messages have different sizes because of the keys.  Deliverability really suffers over 200 bytes so both clients use 200 now as that is still a huge text message and the user experience is much better.

PR #405 (2025-02-06): Limit the length of messages to 200 bytes
Fix #402