meshtastic/web #410: [Bug]: Web Client Bluetooth Configs Not Passing To Device

RevRodge, 2025-02-09:
### Hardware

T-Deck

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.5.20.4c97351

### Description

My T-Deck functioned normally and successfully used Bluetooth/GPS/LoRa running firmware 2.5.18. 

I flashed the firmware to an older one that supported the fancy UI (2.5.3.bfe99b2). After enabling GPS and assigning pins, I verified that those settings were passed through, and the device rebooted without Bluetooth enabled. I opened the web client in Google Chrome, verified that WiFi had been disabled, and enabled Bluetooth. Hit save, and the device reboots, but still without BT. I attempted this process several times.

Assuming that it was an issue with this unsupported firmware, I flashed the latest beta firmware and tried again. I reenabled GPS, made sure WiFi was off, and tried to turn on Bluetooth. The Web Client reports that the configs are saved and the device reboots, but no changes are made on the device. If I close the client and reopen it, it shows that Bluetooth is still disabled.

### Relevant console output

```Shell

```

fifieldt, 2025-02-09:
Bluetooth is not supported on MUI. The MUI uses the "connection" that would normally be used by Bluetooth.

RevRodge, 2025-02-09:
Ah, I understand now. I'm not sure that's what is causing a similar issue in the 2.5.20 version, though. I have two devices running the newest stable, and neither seems to support BT enabling. I re-flashed one back to 2.5.18, and the issue was resolved. I have full config access on the Web Client and can successfully use BT on that device

fifieldt, 2025-02-09:
Fix went in 3 days ago for what you're probably experiencing: https://github.com/meshtastic/firmware/pull/5998

RevRodge, 2025-02-09:
Perfect, thank you for following up

closed 2025-02-09 by RevRodge