meshtastic/web #424: [Bug]: Visual Bug on Map

Hunter275, 2025-02-14:
### Hardware

Heltec V3

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.5.x

### Description

Visual bug on map

[image]

### Relevant console output

```Shell

```

danditomaso, 2025-02-14:
@Hunter275 It will be difficult to handle collision detection on the map. I'd recommend we remove the long names and leave the shortnames, along with adding a drop-shadow and border to create some visual separation between icons.

[image]

Hunter275, 2025-02-14:
@danditomaso 100% yeah we don't need long names

danditomaso, 2025-02-14:
@Hunter275 What do you think of the shadow and border? Should we keep that as well? I'm going to remove the long name in the existing map PR #409

PR #409 (2025-02-18): Fixed map zooming when clicking on map marker. Fixed popup alignment to Map Marker
- Fixed issue where map zoom to arbitrary zoom when clicking on map marker
- Fixed issue with avatar disappearing when clicking on marker
- Fixed popup alignment to marker
- Memoized marker creation with useMemo
- Consolidated marker rendering logic

Fixes #408 
Fixes #424
closed 2025-02-18 by danditomaso