meshtastic/web #427: [Bug]: Screen Turns White Randomly, Requires Refresh (Loses Conversation)

TheAZtech, 2025-02-14:
### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.5.xx

### Description

Description:
After an unknown amount of time, the screen turns completely white. The cause is unclear, but this issue has occurred multiple times over the past week.

Refreshing the page restores functionality, but it returns to the connected devices screen, causing all ongoing conversations to be lost due to the lack of persistence. 

Steps to Reproduce:
	1.	Use the application as normal in Google Chrome.
	2.	After an undetermined period, the screen turns white.
	3.	Refreshing the page brings you back to the connected devices screen.

Expected Behavior:
The app should not crash or turn white unexpectedly. 

Environment:
	•	Browser: Google Chrome (latest version)
	•	OS: macOS Sequoia 15.3

[image]

### Relevant console output

```Shell

```

fifieldt, 2025-02-14:
Thanks for the report! Next time this happens, could you hit F12 and send us a screenshot of the developer console?

danditomaso, 2025-02-16:
@TheAZtech Thanks for reporting this issue!  @fifieldt To provide some additional context during our internal testing we have came across a similar issue to the one reported. So we are aware the issue and we are actively working on a fix (fingers crossed for next week). Due to the severity of it we're aiming to get it merged asap.

booogs, 2025-02-21:
Same issue for me when trying to broadcast via bluetooth on my builds. I never see the message get acknowledged/delivered. Started happening just recently.

booogs, 2025-02-21:
i seemed to have fixed the issue on my end. i found that if you modified your primary channel and gave it a channel name, it won't allow you to broadcoast anything and hits the maximum retransmission, sometimes going to a white screen. once i defaulted the primary channel with a blank channel name, my messages go through and no white screen. hope this helps!

TheAZtech, 2025-03-02:
I _think_ this is what you are looking for? 

[image]

[image]

Hunter275, 2025-03-05:
Fix in #458

closed 2025-03-05 by Hunter275