meshtastic/web #437: How to use the docker command?

alfi4000, 2025-02-19:
I used the docker command on the main page but I was unable to connect to bluetooth or serial!

I tested on both Google Chrome and Edge!

KomelT, 2025-02-19:
You couldn't select the option or it wasn't working?

danditomaso, 2025-02-19:
Was your device connected to your computer using usb C or another power source? Also what kind of Meshtastic device do you have?

alfi4000, 2025-02-19:
Sorry for the unclear explanation it is greyed!

alfi4000, 2025-02-19:
I just connected it to a domain then the tls button switched to the other site with lets encrypt ssl certificate but the Bluetooth and Serial are still greyed!

KomelT, 2025-02-19:
Or you can just use latest hosted version of web ui https://client.meshtastic.org/

closed 2025-02-19 by alfi4000