meshtastic/web #440: [Bug]: Hosted Web client always using http

justinforlenza, 2025-02-20:
### Hardware

Heltec V3

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.5.20.4c97351

### Description

Currently when using the hosted web client, despite the toggle switch being locked to HTTPS, the UI + console show it it still being connected via HTTP, which gets blocked because of mixed content policy

UI Showing `http://`:
[image]

Connection being blocked:
[image]

[image]

### Relevant console output

```Shell

```

danditomaso, 2025-02-20:
@justinforlenza Did the mixed content warnings disappear when connecting using https?

justinforlenza, 2025-02-20:
Yeah when I tested my changes from #442 over HTTPS the mixed content warnings disappeared.

PR #442 (2025-02-20): fix: tls switch for http connections
Fixed the default value for the TLS switch and fixed double state management.

Fixes #440
closed 2025-02-20 by danditomaso