meshtastic/web #447: [Bug]: Config screen turns white

booogs, 2025-02-21:
### Hardware

Other

### Connection Type

Bluetooth

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

na

### Description

Hosted client was working fine in the morning, but now everytime I try to connect to the config screen for any of my builds (rak19007 and heltec mesh node t114), the screen just goes white.

### Relevant console output

```Shell

```

booogs, 2025-02-21:
One thing to note that I just found out is the self-hosted client works perfect in getting into the config, though it will only actually update via serial and not bluetooth.

danditomaso, 2025-02-21:
@booogs @stargatecmd2009 Thanks for reporting this issue, our prod environment had been stuck on an older version of the app that had a bug in the configuration page. Its been resolved after pushing PR #448. You should be able to use the web client as normal now. Please feel free to re-open this issue if you're still experiencing issues.