meshtastic/web #455: [Bug]: Can not scroll up in messages windows.

TheAZtech, 2025-02-22:
### Hardware

Other

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

864f807

### Description

Can not scroll up in messages window. Whatever messages are showing is what you get. Can not scroll up to see past messages. 

[image]

### Relevant console output

```Shell

```

TheAZtech, 2025-02-22:
Here is some better info than my hasty post this morning. In these posted screenshots, the red arrow is pointing to the top of the scroll bar. It is all the way at the top and will not let it go up to see past messages. It does not seem to matter what size the window is. If I zoom the page out I can see the entirety of the message thread. The bottom blue arrow is where the first screenshot is cut off at.

[image]

[image]

danditomaso, 2025-02-28:
@TheAZtech Which version of the app are you using? What is the Commit SHA at the bottom of the navigation bar? I'm wondering if you aren't on the latest version because chats normally start from the bottom of the window and scroll up from there.

PR #487 (2025-03-10): fix: resolved issue with being unable to scroll up in the input field

## Description

This PR fixes an issue with the ChannelChat component that caused an issue when messaged scrolled off the top of the screen you couldn't scroll back to see them. 

## Related Issues

Fixes #455 

## Changes Made

- Moved MessageInput component the MessagesPage level that works for both chat types
- Fixed the layout structure to ensure proper scrolling behaviour
- Better handling of destination and channel props based on active chat type
- Fixed overflow issues for message history
- Added class which allows f [... cut]