meshtastic/web #470: [Bug]: webui 2.5.20: node list not refreshing

spommere, 2025-02-28:
### Category

Other

### Hardware

Other

### Firmware Version

2.5.20beta

### Description

Node is a LilyGo T3-S3. Node was fully erased and reimaged using Webflasher, with the WebUI option.
Connected via BLE to the node, wwitched network config to Wifi, and I can connect w/o problem.

WebUI: Navigation -> Nodes

The list is freshly initialized with existing data from the node, but then never refreshed again.

This afternoon, when I checked, the most recent Last Heard was 19 hours ago, in a list of 41 nodes.
After reloading the page (meshtastic.local), and re-adding the connection to the node,  I had 51 nodes and the top 10 nodes in terms of Last Heard, were within 45 minutes.

Navigating to other menu items such and Config, Map, Messages or anything else, and then back to Nodes, does not refresh or update the node list either. 

I have not used the WebUI extensively, so I can't say if this even worked before.

### Relevant log output

```Shell
NA
```

TheAZtech, 2025-02-28:
I am also experiencing this with both the nodes list and messages. I am running on Chrome with meshtasticd 2.5.22 and 2.5.23 at client.meshtastic.org `593c08f ` Its almost like there is a full disconnect from the firmware somewhere after a period of time. 

[image]

danditomaso, 2025-03-19:
@TheAZtech This should have been fixed by pull request #515. Can you retest and let us know if this is still an issue for you?

ESPnerd, 2025-04-14:
Experiencing similar with client 59a72bc using a Heltec V3 on version v2.6.4.b89355f. Using Chromium Version 135.0.7049.52

PR #671 (2025-06-18): fix: add missing i18n strings

## Description

This PR resolves some missing i18n labels around the UI that were found by @philon-. In one case the Geolocate control had to be disabled as there isnt a straightforward method to internationalise that control. Any labels added will need to be translated in order for the native language version to appear in the UI.  

## Related Issues

Fixes #669 
Fixes #470

## Changes Made

- Refactored timeago to allow for dynamic updating of time ago value
- Added message input placeholder into i18n
- Delivery status tooltip was using the wrong i18n key
- Dyn [... cut]
closed 2025-06-18 by danditomaso