meshtastic/web #481: [Bug]: New connection HTTPs value is wrong

varanauskas, 2025-03-05:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

N/A

### Operating System

macOS 15.2

### Browser

Chrome 133, Safari 18.2

### Expected Behavior

When opening the client via HTTPs it should create new HTTP connections with the "https://" protocol

### Actual Behavior

When trying to create a new connection on a client hosted on HTTPs (https://client.meshtastic.org/), the "Use HTTPS" option is selected and disabled, however, it shows and uses HTTP when trying to connect, resulting in an access error

### Steps to Reproduce

1. Go to https://client.meshtastic.org/
2. Click on "+ New Connection"
3. Ensure "HTTP" tab is selected
4. Observe "IP Address/Hostname" having prefix "http://" instead of "https://"

### Relevant console output

```Shell

```

### Screenshots

[image]

### Additional Context

I think it might be related to commit https://github.com/meshtastic/web/commit/b40079cdc98d6778b6c062a4d33238e839ae9ab5 based on git blame

danditomaso, 2025-03-05:
@varanauskas Sorry about that, this was the result of a bad update deployed to our client.meshtatic.org site. I've patched it now, and the update should be live in about 10 minutes. Can you retry in a few minutes (after refreshing your browser), and see if its still effecting you. Feel free to reopen this issue if you're still experiencing this issue.

closed 2025-03-05 by danditomaso

varanauskas, 2025-03-05:
Thanks for such a quick fix! I've [left a note](https://github.com/meshtastic/web/pull/482#pullrequestreview-2661114036) on how tests could be improved to prevent regressing on this in the future, would it be OK for me to open a PR updating the test with a single line?

PR #482 (2025-03-05): fix: update TLS setting if URL is using HTTPS

## Description

This fix ensures the TLS setting is toggled on if the URL is using HTTPS

## Related Issues

Fixes #481 

## Changes Made

- created variable to track https status 
- updating tls status when site is already using https
commit 39f26f47 (2025-03-05): Merge pull request #482 from danditomaso/issue-481-node-connecting-on-https

fix: update TLS setting if URL is using HTTPS

Fixes #481