meshtastic/web #485: [Bug]: An unhandled error occurred. <Add details here>

ardlen, 2025-03-06:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Heltec V3

### Connection Type

Serial

### Local or Hosted

http://meshtastic.local

### Firmware Version

2.5.20

### Operating System

macos Sonoma 14.6

### Browser

Chrome 

### Expected Behavior

I try to send message in chat

### Actual Behavior

crash page in browser 

### Steps to Reproduce

1) go to Navigation
2)  try to send message in chat

### Relevant console output

```Shell
Error: Minified React error #130; visit https://react.dev/errors/130?args[]=undefined&args[]= for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at Du (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:103111)
    at Bt (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:39025)
    at https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:40005
    at un (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:62117)
    at ey (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:62350)
    at dy (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:76804)
    at Zy (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:116687)
    at NC (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:115765)
    at dp (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:115595)
    at Gy (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:112655)
```

danditomaso, 2025-03-19:
This has been fixed by pull request #516

closed 2025-03-19 by danditomaso

ardlen, 2025-03-20:
Hi, thanks !