meshtastic/web #491: [Bug]: An unhandled error occurred. <Add details here>

admurray256, 2025-03-09:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Heltec V3

### Connection Type

Bluetooth, HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

v2.6.0.f7afa9a

### Operating System

Windows 11, Windows 10

### Browser

Edge

### Expected Behavior

I was configuring after flashing to the alpha firmware, I expected to be able to configure position settings. 

### Actual Behavior

Web client crash. Tried client.meshtastic.org, device's ip address (wifi) with same result. 

### Steps to Reproduce

Not sure how I broke it, I was configuring other things, then tried to go back to the web client. 
This happened using the web UI on device (heltec v3)

### Relevant console output

```Shell
TypeError: Cannot read properties of undefined (reading 'positionFlags')
    at GM (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:473:19025)
    at fh (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:42981)
    at Ih (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:64114)
    at dy (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:74903)
    at Zy (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:116687)
    at NC (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:115765)
    at dp (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:115595)
    at Gy (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:112655)
    at $y (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:122423)
    at ko (https://client.meshtastic.org/assets/index-CbMhZ1eq.js:48:121197)
```

### Screenshots

[image]

admurray256, 2025-03-09:
I was able to recreate the issue. 
1. set wifi for the first time, enter ssid and necessary settings
2. after device restarts, but before the wifi connection is complete, make another change and save again
3. web client freezes while device screen shows wifi information
4. after client refresh, all settings are now bugged. Only way to fix it is compeltely erase and flash again. All the settings pages will be empty: 

[image]

admurray256, 2025-03-09:
I was also able to recreate the issue by purposefully updating settings again via meshtastic CLI

philon-, 2025-08-29:
> [@philon-](https://github.com/philon-) Do you think the recent changes to the config would have fixed this issue? I can retest this tomorrow morning.

It's not obvious to me how we could end up in this state, so I can't say without further testing.

danditomaso, 2025-09-02:
@admurray256 can you retest this using the latest version from HTTPS://client.meshtastic.org ?