meshtastic/web #493: [Bug]: An unhandled error occurred. Minified React error #130

natural1, 2025-03-09:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

T-Deck

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

v2.6.0.0b106d4v2.6.0.0b106d4

### Operating System

Windows 11

### Browser

Chrome 133

### Expected Behavior

Expected display of map 

### Actual Behavior

Map was displaying then it suddenly threw the exception.

### Steps to Reproduce

1. Connect USB cable
2. Browse to client.meshtastic.org
3. Connect via the serial connection dialogue
4. Click on the Map menu on the left.
5. Map will display, then it may throw an exception at some random interval.

### Relevant console output

```Shell
Error: Minified React error #130; visit https://react.dev/errors/130?args[]=undefined&args[]= for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at Du (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:103111)
    at Bt (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:39025)
    at https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:40005
    at un (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:62117)
    at ey (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:62350)
    at dy (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:76804)
    at Zy (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:116687)
    at NC (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:115765)
    at dp (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:115595)
    at Gy (https://client.meshtastic.org/assets/index-B3-8dBuZ.js:48:112655)
```

singlethink, 2025-03-11:
I received the exact same error today while attempting to send a direct message.

## Hardware
T-Echo

## Connection Type
Serial

## Local or Hosted
https://client.meshtastic.org/

## Firmware Version
v2.5.20.4c97351

## Operating System
Ubuntu 24.04.2

## Browser
1.75.181 Chromium: 133.0.6943.141 (Official Build) (64-bit) 

## Expected Behavior
Expected direct message to be sent

## Actual Behavior
Exception returned when I hit the send button

## Steps to Reproduce

Issue does not reproduce reliably.  It may be due to some functionality unrelated to direct messaging.

1. Connect USB cable
1. Browse to client.meshtastic.org
1. Connect via the serial connection dialogue
1. Click "Nodes"
2. Click node in list
3. Type direct message
4. Click send button - sometimes the exception is displayed

danditomaso, 2025-03-19:
This has been fixed by pull request #516

closed 2025-03-19 by danditomaso