meshtastic/web #496: [Bug]: Channel Setting Not Saving.

TheAZtech, 2025-03-11:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

DIY

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

Web UI 3240ac5

### Operating System

macOS Sequoia 15.3.1

### Browser

Chrome 133.0.6943.142

### Expected Behavior

Clicking the Save button should save the modified channel settings.

### Actual Behavior

Setting are not saved when save button is used. 

### Steps to Reproduce

	1.	Navigate to the Channel Settings page.
	2.	Adjust any setting (e.g., channel name, toggles, etc.).
	3.	Click the Save button at the bottom of the page.
	4.	Observe that the settings are not saved.

### Relevant console output

```Shell

```

### Screenshots

[image]

Hunter275, 2025-03-12:
This is most likely a JS issue, investigating.

Hunter275, 2025-03-15:
I've got a fix for this, should have it ready soon.

closed 2025-03-19 by Hunter275