meshtastic/web #501: [Bug]: An unhandled error occurred when sending message to one node via private primary channel

David0669, 2025-03-13:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [ ] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Other

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

v2.5.20.4c97351

### Operating System

Windows 10

### Browser

Edge

### Expected Behavior

Send message to node via primary private channel

### Actual Behavior

Error screen appeared asking to report this error to this GitHub page

### Steps to Reproduce

1 goto Nodes
2. select node on my private primary message channel 
3. Type message
4. Press Send

### Relevant console output

```Shell
Error: Minified React error #130; visit https://react.dev/errors/130?args[]=undefined&args[]= for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at Iu (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:103111)
    at Bt (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:39025)
    at https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:40005
    at un (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:62117)
    at ny (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:62350)
    at hy (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:76804)
    at Xy (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:116687)
    at NC (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:115765)
    at dp (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:115595)
    at Py (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:112655)
```

danditomaso, 2025-03-13:
Thanks for letting us know about this, we have a bug fix in progress to resolve this issue, it should be in place by this weekend.

danditomaso, 2025-03-19:
This has been fixed by pull request #516

closed 2025-03-19 by danditomaso