meshtastic/web #506: [Bug]: An unhandled error occurred. When closing the "save keys" popup

RCGV1, 2025-03-14:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [ ] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Not Applicable

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.6.1

### Operating System

Mac OS

### Browser

Arc Browser (Chromium)

### Expected Behavior

Just sent a direct message, and the " save keys popup occurred,d so I closed it and got the error.

### Actual Behavior

Crashed

### Steps to Reproduce

1. Send direct message
2. Immedietly close save keys popup

### Relevant console output

```Shell
Error: Minified React error #130; visit https://react.dev/errors/130?args[]=undefined&args[]= for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at Iu (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:103111)
    at Bt (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:39025)
    at https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:40005
    at un (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:62117)
    at ny (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:62350)
    at hy (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:76804)
    at Xy (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:116687)
    at NC (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:115765)
    at dp (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:115595)
    at Py (https://client.meshtastic.org/assets/index-ZhfJGcPW.js:48:112655)
```

RCGV1, 2025-03-14:
UPDATE: It crashed when I send direct message, unrelated to the popup

danditomaso, 2025-03-14:
@RCGV1 thanks for the detailed report. I'll pause my work on the queueing and find a better way  to better handle any of the no channel errors we are getting on the front-end

closed 2025-03-19 by danditomaso