meshtastic/web #510: [Bug]:

adokitkat, 2025-03-16:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Other

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.6.1.7c3edde

### Operating System

Windwos 11

### Browser

Chrome 134

### Expected Behavior

Be able to use the web UI client with multiple devices connected.

### Actual Behavior

The first connected device connects OK, the second one does not and just shows up as unknown. Console log output is below,.

### Steps to Reproduce

Just try to connect to 2 device via serial connection in the same browser tab. Using 2 tabs, 1 for each device works fine though.

### Relevant console output

```Shell
index-ZhfJGcPW.js:512 Uncaught (in promise) TypeError: Failed to execute 'pipeTo' on 'ReadableStream': Cannot pipe a locked stream
    at new af (index-ZhfJGcPW.js:512:71329)
    at af.createFromPort (index-ZhfJGcPW.js:512:71616)
    at async u (index-ZhfJGcPW.js:512:72183)
    at async onClick (index-ZhfJGcPW.js:512:72606)

02:40:00:173    INFO    [iMeshDevice]    HandleFromRadio 📱 Received Node Info packet for node: 1039968503
index-ZhfJGcPW.js:67 02:40:00:196    WARN    [iMeshDevice]    HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
index-ZhfJGcPW.js:67 02:40:00:208    WARN    [iMeshDevice]

...

HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
index-ZhfJGcPW.js:67 02:40:00:408    WARN    [iMeshDevice]    HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
index-ZhfJGcPW.js:67 02:40:00:417    ERROR    [iMeshDevice]    HandleFromRadio ❌ Invalid config id received from device, expected 87052930 but received 533338488

index-ZhfJGcPW.js:78 Packet 120725696 of type wantConfigId timed out
Uncaught (in promise) {id: 120725696, error: 3}
index-ZhfJGcPW.js:78 Packet 299154254 of type wantConfigId timed out
Uncaught (in promise) {id: 299154254, error: 3}
```

danditomaso, 2025-03-22:
There is some discussion in the firmware discord channel about getting this feature added and it isn't certain the firmware will allow for multiple connections. Fingers are crossed for it to be included as its been requested by multiple people here. You may not see the feature being available for a few weeks to a month though.

bogdanbrudiu, 2025-04-01:
Can you please point me to the discord discussion? I am not able to find it...

danditomaso, 2025-05-25:
At this time its not possible to connect two devices to the same WebUI. We may revisit this at a later time but for now its a limitation of the WebUI.

closed 2025-05-25 by danditomaso