meshtastic/web #523: [Feature Request]: Support toggling UDP Multicast

vidplace7, 2025-03-18:
### Prerequisites

- [x] I have searched existing issues to ensure this feature hasn't already been requested
- [x] I have checked the documentation to verify this feature doesn't already exist

### Problem Statement

Please add support for toggling the new "UDP Multicast" feature added in
- https://github.com/meshtastic/firmware/pull/5779
- https://github.com/meshtastic/protobufs/pull/643

Currently can only be enabled via Python with this config:
```sh
meshtastic ... --set network.enabled_protocols 1
```

Should be a boolean value to enable/disable, I imagine.

### Proposed Solution

Add true/false selector for "Enable mesh over UDP"

In Android it will be presented like this:
[image]

### Importance

Nice to have

danditomaso, 2025-03-28:
@vidplace7 If UDP via mesh is enabled, does anything else in the network stack need to be disabled?

vidplace7, 2025-03-28:
> @vidplace7 If UDP via mesh is enabled, does anything else in the network stack need to be disabled? 

Nope, Mesh via UDP can be toggled independently of other settings.

PR #547 (2025-03-31): feat: Add udp over mesh toggle

## Description

This PR incorporates the additional functionality of UDP over Mesh, which allows for Mesh data to be sent to another node using UDP instead of sending to the Mesh (I think). The toggle was added and connected to the proper Protobuf enum. Additionally, Zod was replaced in the Network config  instead of `class-variance-authority`, as Zod is 1/5th the size of the other library this is the first of many validation configs that will be refactored with Zod.  

## Related Issues

Fixes #523 

## Changes Made

- Added udp over mesh toggle and connected t [... cut]
closed 2025-03-31 by danditomaso