meshtastic/web #530: [Bug]: Custom Channels

Sprout004, 2025-03-21:
### Category

Other, Serial

### Hardware

T-Deck

### Firmware Version

2.6.2

### Description

Unable to enter/save custom channel info through the WebUI. When inputting channel info on the actual device, when the device is rebooted the channels are removed. 

It does work when connecting to the device via BlueTooth, setting everything up and then disconnecting. 

### Relevant log output

```Shell

```

thebentern, 2025-03-21:
Hi, We will need some serial logs to see what is going on here. You can use the serial monior on https://flasher.meshtastic.org or the one built into the python CLI with `meshtastic --noproto`.

Sprout004, 2025-03-25:
Hey man, sorry for the delay. 

More than happy to get you what you want, just not sure how to go about it as I am a bit noob…

I see serial monitor through the webflasher, but when that is running and I go to client.meshtastic.org the device doesn’t come up…

mikerocklewitz, 2025-03-29:
I can confirm this is not unique to the tdeck.

I have not been able to edit channel settings from the web client over bluetooth or serial on any device.  The changes don't get passed to the module

mikerocklewitz, 2025-04-01:
Doesn't seem fixed for me.  3 web browsers. 2 nrf based devices.  Newest 2.6.4. Serial or Bluetooth is irrelevant.

Trying to enable or edit any secondary channel, the submit button does nothing.  If I click on messages under navigation and go back to channels all changes are gone.

It's like the submit button under channels isn't connected to any functionality.

The save button under all things "config" works fine.

Only time I've gotten the channel settings page submit button to work is with spinning up a docker container and doing it over wifi to a heltec v3, but it's been a few weeks and I can't remember the versions I was experimenting with....my permanent node is nrf based......

Is there a way to bluetooth or serial from the docker container?

Sprout004, 2025-04-02:
It is still not working through the web client. When I put a channel in, nothing happens. 

The ones entered via Bluetooth and the app are still there.

mikerocklewitz, 2025-04-02:
I'm not sure how I can go about giving console data when either connection ties up the serial port.

 Would i have to connect seperate uart?  I have everything possible to make anything happen here, just not sure of the methodology

Sprout004, 2025-04-02:
Also, not sure there are any errors generated, you click submit on the channel page and nothing happens. No "it didn't work" or "it did work" just nothing. If you navigate away from the channel page, thinking the info you entered is saved,and come back to it, the channel info is blank, like you never entered anything in the first place.

mikerocklewitz, 2025-04-02:
> Also, not sure there are any errors generated, you click submit on the channel page and nothing happens. No "it didn't work" or "it did work" just nothing. If you navigate away from the channel [...]

This exactly

danditomaso, 2025-04-07:
@Sprout004 This should have been fixed and is now on live. Please retest and feel free to re-open this issue if you find it still exists.

closed 2025-04-07 by danditomaso

mikerocklewitz, 2025-04-07:
Success!

Sprout004, 2025-04-08:
Fixed!!

I think Mike and I might be best friends now!