meshtastic/web #535: client.meshtastic.org

pdm0393, 2025-03-24:
I really like the web client i'm missing in fact two things when using map. It would be great if you could filter only active clients is see on t-deck for example 21 active but count 40 nodes on map. and it would be great if you could click on node on map and then driectly go to send direct message

regards, 
peter

James9074, 2025-03-27:
Hey @pdm0393 - I've opened a PR for this feature: https://github.com/meshtastic/web/pull/540

danditomaso, 2025-03-29:
@pdm0393 Thanks for @James9074 for getting this feature added. On https://client-test.meshtastic.org you will see this feature. We are releasing a 2.6 of the web client in the next week so you should be this being available.

danditomaso, 2025-03-29:
Closing this issue now that the code has been deployed to the test web client.

closed 2025-03-29 by danditomaso

pdm0393, 2025-03-29:
Hi, I'm ver curious when it will be released. I think people wil like it if they can just click 😁I'm a nono whennit is about github and how it works but i think you guys did a great job. Wil it also be available when i connect directlybto the device because with https:// you cannot connect to device over wifi. But when i connect to ip adress of device it starts in my opinion an other version of webclient. Is that correctRegards, Met vriendelijk groet,Peter de Monyé

James9074, 2025-03-29:
@pdm0393 It shouldn't matter which method you use to connect to the device, the map will appear identical. Look for a release within the next week or two here: https://github.com/meshtastic/web/releases

pdm0393, 2025-03-29:
Great work guys realy thank you👍👏👏Met vriendelijke groet,Peter de Monyé