meshtastic/web #537: [Bug]: An unhandled error occurred. Crashed on WiFi connection

snwvo7, 2025-03-25:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Heltec V3, Heltec Wireless Tracker

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

v2.5.20.4c97351

### Operating System

Windows 11

### Browser

Microsoft Edge

### Expected Behavior

Try to connect via WiFi http (ipadress), before I did work for 15 minutes with the same hard/software

### Actual Behavior

connected the Heltec via wifi it was loading messages when it crashed.
I have to reboot the Heltec to be able to connect again but same issue appreared

### Steps to Reproduce

1) Open Meshtastic Web
2) New Connection
3) HTTP + [IpAdress]
4) messages loading ----- hang
5) connection lost

### Relevant console output

```Shell
TypeError: Cannot read properties of undefined (reading 'positionFlags')
    at QM (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:477:19330)
    at fh (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:42981)
    at Ih (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:64114)
    at py (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:74903)
    at Xy (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:116687)
    at wC (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:115765)
    at dp (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:115595)
    at Py (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:112655)
    at t1 (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:122423)
    at Vo (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:121197)
```

### Additional Context

I did move the Heltec closer to the WiFi router but still the same issue.

danditomaso, 2025-03-25:
@snwvo7 Can you try this copy of the Meshtastic web client? We have a bug fixed deployed but we need to verify it fixes your issue. 

https://web-test-9iwc6z4ek-meshtastic.vercel.app/

snwvo7, 2025-03-25:
Testing is possible, how do you want to move ahead?

https://web-test-9iwc6z4ek-meshtastic.vercel.app/

https://web-test-9iwc6z4ek-meshtastic.vercel.app/

danditomaso, 2025-03-25:
@snwvo7 Thanks. If you can test using the web clienthttps://web-test-9iwc6z4ek-meshtastic.vercel.app/. Its a copy of our live site except with the bug fix added. Its a temporary site to preview if this fix resolves your site, and will go away once the code is merged. Can you try connecting to your device from that web client and let me know if you run into the same problem you reported above?

closed 2025-03-26 by danditomaso

snwvo7, 2025-03-28:
Hi Dan,

I am not able to connect via WiFi.
I have tried it from Windows 11 and MacOs.
I have also used 2 modules,

  1.
Heltec Wireles Tracker
  2.
Heltec V3

After adding the ip-adres and push the connect button it stays in connecting.

Best Regards. Martin.

Closed #537<https://github.com/meshtastic/web/issues/537> as completed via #538<https://github.com/meshtastic/web/pull/538>.

PR #538 (2025-03-26): fix: ensured undefined position flags are handled

## Description

This adds an undefined check to the configuration position flags. It avoids the case where a nil is returned instead of a valid integer.

Fixes #537 Fixes #504 

## Additional Notes