meshtastic/web #539: [Bug]: An unhandled error occurred. <Add details here>

OnyxtheDragon, 2025-03-27:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Other

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

v2.5.20.4c97351

### Operating System

Windows 11

### Browser

Chrome 134.0.6998.166

### Expected Behavior

Settings populating with their configured values, key backup working successfully

### Actual Behavior

I was attempting to back up my keys before updating to the latest alpha for testing. After pressing the backup keys button nothing happened. i rebooted the device multiple times and the settings no longer populated inside the web client

I was attempting to configure my T1000-E using the web client to back up my keys for a firmware update. none of the settings values loaded and the error appeared after 1-2 minutes

### Steps to Reproduce

Attempt to connect to T1000-E over the web client, try to back up the keys and nothing happens

### Relevant console output

```Shell
TypeError: Cannot read properties of undefined (reading 'positionFlags')
    at QM (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:477:19330)
    at fh (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:42981)
    at Ih (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:64114)
    at py (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:74903)
    at Xy (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:116687)
    at wC (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:115765)
    at dp (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:115595)
    at Py (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:112655)
    at t1 (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:122423)
    at Vo (https://client.meshtastic.org/assets/index-BjuPOQ8t.js:48:121197)
```

danditomaso, 2025-04-05:
@OnyxtheDragon Can you test again? I had pushed some updates to https://client-test.meshtastic.org that should solve your problem. This will be pushed to https://client.meshtastic.org in our May release window.

OnyxtheDragon, 2025-04-14:
Issue appears to be resolved!

Tested on latest firmware using T1000-E, alongside the original device.

Thanks for the fix!

closed 2025-04-14 by OnyxtheDragon