meshtastic/web #545: [Bug]: meshtastic/web nighty docker build does not start at all

dirksan28, 2025-03-29:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

irrelevant

### Operating System

Linux raspi3 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

### Browser

wget

### Expected Behavior

I expect to be able to connect to the local hosted docker release of the web client.
This works well for latest release (5a142e6) ~5 month old. 

Example:
```bash
dirk@raspi3:~                                                                                                                                                                                       
▶ docker run -d -p 8080:8080 --restart always --name Meshtastic-Web ghcr.io/meshtastic/web        
Unable to find image 'ghcr.io/meshtastic/web:latest' locally
latest: Pulling from meshtastic/web
f91f19baa623: Pull complete 
f6dbf5241cf6: Pull complete 
cd3dafff381f: Pull complete 
14463fb3d0f5: Pull complete 
Digest: sha256:ef572f994c7189f67187431ea75bfb9e3ecf9601c6a693f47ed4e1accc3a77a4
Status: Downloaded newer image for ghcr.io/meshtastic/web:latest
9ad2179e9e5940bbe115b0bd823f7c7cdf02440b169ce85e758f2de4b54e013e

dirk@raspi3:~                                                                                                                                                                                       
▶ wget localhost:8080
--2025-03-29 13:35:53--  http://localhost:8080/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:8080... connected.
HTTP request sent, awaiting response... 200 OK
Length: 492 [text/html]
Saving to: ‘index.html’

index.html                                        100%[==========================================================================================================>]     492  --.-KB/s    in 0s      

2025-03-29 13:35:53 (11.6 MB/s) - ‘index.html’ saved [492/492]
```

Unfortunately when I try to run the nightly image (```ghcr.io/meshtastic/web:nightly```) I'm not even able to connect to the Ngix instance running inside the docker container. 

### Actual Behavior

Here's what I get instead:

```bash
dirk@raspi3:~                                                                                                                                                                                       
▶ docker run -d -p 8080:8080 --restart always --name Meshtastic-Web ghcr.io/meshtastic/web:nightly
Unable to find image 'ghcr.io/meshtastic/web:nightly' locally
nightly: Pulling from meshtastic/web
30dc6f8a69df: Pull complete 
a8838b5a70b2: Pull complete 
e84b06e3282c: Pull complete 
8a1143046b1c: Pull complete 
7a33d50f4204: Pull complete 
9da97b51dd1e: Pull complete 
a68d10edc078: Pull complete 
c46b11f359eb: Pull complete 
bb4fcde4e373: Pull complete 
Digest: sha256:732cfec29b6ec976d97cfeaf81e42af08b1cf83559e45c27dab61319e65ae55a
Status: Downloaded newer image for ghcr.io/meshtastic/web:nightly
c99cebdcda1301feb8dfeab7b5946fd82e28eabbd37a8a2226da9fbf24488fcd

dirk@raspi3:~                                                                                                                                                                                       
▶ wget localhost:8080
--2025-03-29 13:31:16--  http://localhost:8080/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:8080... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

--2025-03-29 13:31:17--  (try: 2)  http://localhost:8080/
Connecting to localhost (localhost)|::1|:8080... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

--2025-03-29 13:31:19--  (try: 3)  http://localhost:8080/
Connecting to localhost (localhost)|::1|:8080... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

^C
```

### Steps to Reproduce

1. docker run -d -p 8080:8080 --restart always --name Meshtastic-Web ghcr.io/meshtastic/web:nightly
2. wget localhost:8080

### Relevant console output

```Shell

```

dirksan28, 2025-03-29:
Wow, that was quick! Your signing worked, the client is now reachable. 
I'm excited to try out all the new features. 

Thanks @danditomaso for the help and the fix.

closed 2025-03-29 by dirksan28

danditomaso, 2025-03-29:
Happy to help, do me a favour and try pulling tonight/tomorrow's nightly build to ensure going forwards docker builds work going forwards.

dirksan28, 2025-03-30:
> Happy to help, do me a favour and try pulling tonight/tomorrow's nightly build to ensure going forwards docker builds work going forwards.

Just pulled the current nightly  ([4d30558acaefac87c023158d1b9db0a72b00c7aa](https://github.com/orgs/meshtastic/packages/container/web/384671035?tag=4d30558acaefac87c023158d1b9db0a72b00c7aa)) which now runs perfectly.

Thanks again.

PR #546 (2025-03-29): Fixed Docker Builds Not Serving Site on 8080

## Description

This PR contains fixes for the docker build process which is used with the nightly and release builds. Previously when the Docker container was built the site wasn't available on port 8080, this is now being configured based on the config file passed to nginx from the `./infra` folder. 

## Related Issues

Fixes #545 

## Changes Made

- Added external configuration file for nginx
- Added gzip compression to docker build
- Updated Nightly and Release github workflows

## Testing Done

Manual testing done locally via docker desktp.