meshtastic/web #575: [Bug]: Direct Messages not displaying in webui 2.6.4

roberthadow, 2025-04-15:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Heltec V3

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

v2.6.4.b89355f

### Operating System

Ubuntu 24.04.2 LTS

### Browser

Version 134.0.6998.165 (Official Build) (64-bit)

### Expected Behavior

I have four Heltec V3s running in my shack. I exercise/test them with CLI scripts over WiFi.  Those scripts include broadcast and direct messaging, over primary and scondary channels, with and without acknowledgement.  

I keep four tabs open in my browser that reliably display broadcast messages from each device.  I launch my test scripts in a terminal, where I see acknowledgements (when the command calls for them).

Bash lines like this

```
meshtastic --host 192.168.1.200 --dest '!b03bb4f8' --sendtext 'Hello from BB38 W2NJ' 
sleep 30s
meshtastic --host 192.168.1.211 --ch-index 1 --dest '!b03dbb38' --sendtext 'Hello from CE44 W2NJ w ack' --ack
sleep 30s
meshtastic --host 192.168.1.200 --dest 2956853972 --sendtext 'DM Int from BB38 W2NJ w ack' --ack
sleep 30s
```
generate radio messages that display fine on the device OLED and my Android phone connected by USB-C. They are never served up and visible in the browser.

### Actual Behavior

No direct messages addressed to a single device, either by byte address or integer address ever paint in the brownser. They display more or less reliably on the device OLED or over the serial connection to an Android device.

### Steps to Reproduce

(1) make a bash script like this (see above for my particulars):

```
meshtastic --host <IP> --dest <byte> --sendtext 'Hello from BB38 W2NJ' 
sleep 30s
meshtastic --host <IP> --ch-index 1 --dest <byte> --sendtext 'Hello from CE44 W2NJ w ack' --ack
sleep 30s
meshtastic --host <IP> --dest <int> --sendtext 'DM int from BB38 W2NJ w ack' --ack
sleep 30s
```
(2) run it. Observe acknowledgments.

(3) see if the messages display on the OLED and phone

(4) see if the messages display in browser

[... 339 lines cut ...]
  "detectionSensor": {
    "minimumBroadcastSecs": 45,
    "detectionTriggerType": "LOGIC_HIGH",
    "enabled": false,
    "stateBroadcastSecs": 0,
    "sendBell": false,
    "name": "",
    "monitorPin": 0,
    "usePullup": false
  },
  "paxcounter": {
    "enabled": false,
    "paxcounterUpdateInterval": 0,
    "wifiThreshold": 0,
    "bleThreshold": 0
  },
  "version": 0
}

Channels:
  Index 0: PRIMARY psk=default { "psk": "AQ==", "moduleSettings": { "positionPrecision": 14, "isClientMuted": false }, "channelNum": 0, "name": "", "id": 0, "uplinkEnabled": false, "downlinkEnabled": false }
  Index 1: SECONDARY psk=secret { "psk": "6A96qbe0orBDenvilleCERT2PWlN+MpYRyRuB2xeFEA=", "name": "Denville CR", "moduleSettings": { "positionPrecision": 32, "isClientMuted": false }, "channelNum": 0, "id": 0, "uplinkEnabled": false, "downlinkEnabled": false }

Primary channel URL: https://meshtastic.org/e/#CgcSAQE6AggOEgwIATgBQANIAVAeaAE
Complete URL (includes all channels): https://meshtastic.org/e/#CgcSAQE6AggOCjMSIOgPeqm3tKKwQ3p74pZXghEU9j1pTfjKWEckbgdsXhRAGgtEZW52aWxsZSBDUjoCCCASDAgBOAFAA0gBUB5oAQ

roberthadow, 2025-05-03:
### Observations as of 3 May 
From / To | Message type     | Bare device | Android   | IOS | Webui  | CLI    |
:------------|:-----------------------|:-----------------|:-------------|:------|:----------|:--------|
Android   | broadcast [^all]   | tested ok      | tested ok | NT  | tested ok| NA   |
Android   | direct message   | tested ok   | tested ok | NT  | fails[4]| NS    |
Android   | admin            | tested ok   | NT        | NA  | NA     | NA     |
IOS       | broadcast [^all] | NT          | NT        | NA  | NT     | NA     |
IOS       | direct message   | NT          | NT        | NA  | NT     | NA     |
IOS       | admin            | NT          | NT        | NA  | NT     | NA     |
Webui     | broadcast [^all] | tested ok   | tested ok | NT  | tested ok| NS   |
Webui     | direct message   | NS          | NS        | NT  | NS     | NS     |
Webui     | admin            | tested ok   | NA        | NT  | NA     | NS     |
CLI       | broadcast [^all] | tested ok   | tested ok | NT  | tested ok | NS  |
CLI       | direct message   | tested ok   | fails [1] | NT  | fails[5] | NS   |
CLI       | admin over USB   | tested ok   | NA        | NA  | NA     | NA     |
CLI       | admin over WiFi  | tested ok   | NA        | NA  | NA     | NA     |
CLI       | admin over LoRa  | fails[2]    | NA        | NA  | NA     | NA     |

***Notes***
* NA not applicable
* NS not supported
* NT not tested

***Test conditions***
* nodeDB fully populated; no missing data, no errors
* webui -- static IP local WiFi; all browser windows recently refreshed
* CLI messages no less than 51 second pause between commands

***Results***
* fails[1] -- ack received; notification appears on Android; message not reliably displayed on Android; nodeDB updates with received time, though
* fails[2] -- PKI error
* fails[4] -- sent indication; does not show on Android display
* fails[5] -- even with ack, messages don't show on webui

philon-, 2025-05-09:
I have tried using [client-test Web UI](http://client-test.meshtastic.org) and can't replicate this on

WebUI <-> WebUI
WebUI <-> iOS
WebUI <-> Android (emulated, connection over IP)

Do you have any other client connected to that same node when you're trying to use the Web UI? 

Also, you can try opening developer tools / web inspector and see what the console output is - all packages received in the UI get logged. Should show up something similar to 

```index-CmXB4QUZ.js:67 09:25:55:943	TRACE	[iMeshDevice:HttpConnection]	HandleMeshPacket 📦 Received TEXT_MESSAGE_APP packet```

danditomaso, 2025-06-12:
Closing as we are unable to reproduce.

kikkia, 2025-06-16:
Question as I have this issue to and was wondering about a detail. 
I also had my phone connected to my permanent home node with the meshtastic app and my phone was receiving the messages but not the web interface on my desktop. However if I left the house, the messages go to the desktop, until my phone connected again, then they go to the app on the phone. 

Does the device delete received messages after any client grabs them to free up the memory? Is it possible that my phone is grabbing the messages before my webui on my desktop can, so by the time by desktops web ui look they are free'd?

philon-, 2025-06-16:
> [@danditomaso](https://github.com/danditomaso) Question as I have this issue to and was wondering about a detail. I also had my phone connected to my permanent home node with the meshtastic app and my phone was receiving the messages but not the web interface on my desktop. However if I left the house, the messages go to the desktop, until my phone connected again, then they go to the app on the phone.
> 
> Does the device delete received messages after any client grabs them to free up the memory? Is it possible that my phone is grabbing the messages before my webui on my desktop can, so by the time by desktops web ui look they are free'd?

The device firmware is only able to serve one client at a time. This is a firmware restriction and not a limitation in the web client.

philon-, 2025-06-16:
> Both of the clients were setup to use it over http. when messages are fetched over http does that clear them off of the device?

Yes, as soon as they are sent to the client they are removed from the device.

If you were using a networked connection with the phone (Android) app, the connection is not actually over HTTP but TCP. Either way, the unfortunate case is that the firmware is not equipped to handle more than one concurrent client connection, regardless of transport (HTTP, TCP, Serial, Bluetooth).