meshtastic/web #578: [Bug]: Chrome (on Linux): "This connection type requires Web Bluetooth. Please use a supported browser, like Chrome or Edge."

sanderjo, 2025-04-20:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Heltec V3

### Connection Type

Bluetooth

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.5.16

### Operating System

ubuntu 25.10

### Browser

Chrome 135.0.7049.95 (Official Build) (64-bit)

### Expected Behavior

Within Chrome (on Linux), I want to connect to my Heltec via Bluetooth

### Actual Behavior

With Chrome, via https://client.meshtastic.org/, -> Bluetooh, it says "This connection type requires Web Bluetooth. Please use a supported browser, like Chrome or Edge.". 

That's weird, because I do use Chrome

[image]

### Steps to Reproduce

With Chrome, via https://client.meshtastic.org/, -> Bluetooh, it says "This connection type requires Web Bluetooth. Please use a supported browser, like Chrome or Edge.". 

### Relevant console output

```Shell
Nothing on cosole (will recheck)
```

### Additional Context

The bad thing: from Ubuntu, I do scan Bluetooth for devices, but it doesn't find the Heltec

garthvh, 2025-04-20:
Do you have experimental web platform features enabled?

sanderjo, 2025-04-20:
On another Ubuntu system, immediatly the same error. And I cannot click on new device.

[image]

On Ubuntu level, I can connect to dho1 (one of my two Heltecs)

[image]

sanderjo, 2025-04-20:
> Do you have experimental web platform features enabled?

In Chrome?

So I must download Chrome Beta for that ... ?

[image]

sanderjo, 2025-04-20:
FWIW: on Windows 11, I can connect, but then ... nothing

[image]

[image]

fifieldt, 2025-04-20:
> So I must download Chrome Beta for that ... ?

chrome://flags

closed 2025-04-22 by Hunter275

kr4k0n, 2025-08-27:
Tried with Linux Debian 6.2 as well with Brave Browser v1.81.136 (Aug 20, 2025) which was built on Chrome, and "Experimental Web Platform features" enabled. Useless meshtastic web client feature = Nothing. Clicking "New Device" does nothing. T1000-e bluetooth connected to PC. Phone connects fine not web client. Windows 11 latest does not even detect the card Bluetooth. Just strip that feature off the web client if it's **not planned**. It's giving everyone false hopes. >_> .. Serial connection doesn't even have internet passthru..