meshtastic/web #579: [Bug]: Web Console Interface is Missing

AA7US, 2025-04-21:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Other

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

v2.6.4.b89355f

### Operating System

Mac OS 15.4.1

### Browser

Opera, Safari, Chrome

### Expected Behavior

Entering a WiFi enabled node's IP address or http://meshtastic.local/ into a browser on same LAN should give access to node's web console for configuration and operation.

### Actual Behavior

Node reports:

"Web server is running.

The content you are looking for can't be found. Please see: FAQ

admin"

Node CAN still be accessed using a serial connection through https://client.meshtastic.org/.

### Steps to Reproduce

1. Enable WiFi on node using iOS app.
2. After node reboots, enter node's IP address in browser on same LAN. 

### Relevant console output

```Shell

```

### Screenshots

[image]

### Additional Context

This bug was observed on a Station G2 node (ESP32-s3).

This bug appears to be the same as reported here: https://github.com/meshtastic/web/issues/333

The node's web console was working fine after flashing between firmware versions 2.5.20 and 2.6.4 until I did a "Full erase and install" of version 2.6.4, then it stopped working.

AA7US, 2025-04-21:
I just reflashed version 2.6.4 again WITHOUT an erase first and still no web console.

fifieldt, 2025-04-21:
In the flasher, did you hit "full erase" and then "include webserver"?

AA7US, 2025-04-21:
@fifieldt Thanks for the suggestion. It looks like it's now called "Bundle Web UI" and I'll give that a try and report back.

[image]

AA7US, 2025-04-21:
@fifieldt Enabling "Bundle Web UI" before flashing did the trick!  Thanks again.

closed 2025-04-21 by AA7US

jekeam, 2025-08-06:
> [@fifieldt](https://github.com/fifieldt) Enabling "Bundle Web UI" before flashing did the trick! Thanks again.

Please tell me why I don't have the option to choose "Bundle Web UI"?

[image]