meshtastic/web #584: Why are only ESP32s supported for HTTP?

aeblyve, 2025-04-25:
I will start off by saying this isn't a complaint: I'm interested to know. 

I'm able to connect to my RAK4631 using the Meshtastic CLI over TCP/IP.

Is there any hard limitation that prevents the implementation of HTTP for this device? Or perhaps the web client could be made to work over the lower-level TCP/IP protocol?

garthvh, 2025-04-25:
NRF does not have a wifi chip

closed 2025-04-25 by garthvh

aeblyve, 2025-04-25:
It does have ethernet support, though, with an add-on.

garthvh, 2025-04-25:
> It does have ethernet support, though, with an add-on card.

For MQTT mainly, there is no web server on the device, TCP works in a narrow way but not the complete HTTP api

aeblyve, 2025-04-25:
So there's no hard reason why it's not possible, it's just not implemented, or easy to implement quickly?

garthvh, 2025-04-25:
> So there's no hard reason why it's not possible, it's just not implemented, or easy to implement quickly?

There is no web server or http endpoints, and a NRF chip has considerably less resources than an esp32 device, non trivial.