meshtastic/web #614: [Feature Request]: Web UI simple authentication

bmatic385, 2025-05-08:
### Prerequisites

- [x] I have searched existing issues to ensure this feature hasn't already been requested
- [x] I have checked the documentation to verify this feature doesn't already exist

### Problem Statement

I run a Meshtastic node on my roof, connected via ethernet to my LAN, then do a port forward to Meshtastic node Web UI to connect to it over the internet when I am at work, if I do that now, everyone can access Web UI on my node.

### Proposed Solution

It would be nice if simple authentication is implemented in Web UI which is served directly from an node.

### Importance

Nice to have

fifieldt, 2025-05-08:
Hi! As a project, we strongly recommend not to connect Meshtastic devices to untrusted networks, like the internet. Its not convenient, I know, but the firmware is developed with the assumption that this isn't done, so there could be security badness :) 

If you do want to have the Web UI available externally, a couple of suggestions are: to run an authenticating proxy in front of the device, and/or access the device using a VPN.

danditomaso, 2025-05-08:
Yeah something like TailScale would be a perfect fit here. You install it on your local machine at home, install it at work and you have a safe/secure way to access your lan and mesh device while at work. 

Unfortunately Authentication of any kind won't be added to the webUI, it's too easy to add a bad implementation and have people think they are secure. For that reason and many more this isn't on our roadmap.

closed 2025-05-08 by danditomaso