meshtastic/web #624: [Bug]: Map Avatar shows unexpected newline

philon-, 2025-05-21:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

-

### Operating System

Fedora 41

### Browser

Firefox

### Expected Behavior

Node Avatar displays the same way everywhere on the client, including map. 

### Actual Behavior

Nodes with a hyphen in the short name sometimes render the Avatar with a newline.

Does not happen for every Avatar with a hyphen in the short name.

### Steps to Reproduce

1. Open map interface
2. Search for node with hyphen
3. ??

### Relevant console output

```Shell

```

### Screenshots

[image]

[image]

danditomaso, 2025-05-25:
@philon- I wonder if its because in that top screenshot there is a black border around the icon and the focus is the one causing the text to overflow?

philon-, 2025-05-26:
Doesn't seem like it is affected by the focus. 

This is also happening on my macOS device, using Safari, although for more nodes(!).

No extra characters are present in the shortName.

[image]

commit 8413f634 (2025-05-27): Fix #624 (#631)
closed 2025-05-27 by danditomaso