meshtastic/web #633: [Bug]: Saving device Configurations with web client on Linux

dword4, 2025-05-27:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Heltec V3

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.6.4.b89355f

### Operating System

Ubuntu 25.04 x86_64

### Browser

Firefox 138.0.4, Chromium 136.0.7103.113 (snap)

### Expected Behavior

After changing a settting such as disabling Wifi (when connected with HTTP) it should drop the connection after a reboot and not be able to reconnect with HTTP

### Actual Behavior

The client proceeds like it is saving and when the device reloads the setting has not been altered.

### Steps to Reproduce

1. Change a setting through web client using HTTP (disable Wifi for ex)
2. Click Save
3. After reset attempt to connect back with HTTP and it connects right back

### Relevant console output

```Shell

```

roberthadow, 2025-07-01:
My heuristic is this:  If I intend to change IP configuration, do it any other way than CLI-WiFi.  If I intend to change IP configuration, I use CLI-USB.  I don't bother changing configuration using Android; it leaves me no record and no way to easily repeat the process.

Likewise: If I intend to change BLE configuration, I do it using CLI-WiFi.

fifieldt, 2025-07-30:
I think we fixed this in firmware: https://github.com/meshtastic/firmware/pull/7484

Many thanks for the highly detailed triage. That really helped lock down the problem.