meshtastic/web #668: [Bug]: Number of nodes mismatch

philon-, 2025-06-17:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Not Applicable

### Connection Type

HTTP

### Local or Hosted

http://meshtastic.local

### Firmware Version

-

### Operating System

-

### Expected Behavior

Number of nodes in sidebar should match number of nodes shown in node list

### Actual Behavior

Node list filters out nodes, resulting in a higher number in the sidebar than the node list.

### Steps to Reproduce

Have sufficiently large database so that there are nodes without nodeinfo

### Relevant console output

```Shell

```

### Screenshots

[image]

Mictronics, 2025-08-30:
I can confirm that issue being related to the filter applied in getNodes.
My nodes count showing 301 but the node table itself contains only 275 rows. 
Searching for a specific node by name returns an empty list.
[image]
The same search in the message node list returns the correct result:
[image]
Even when the node is makred favorite it's not in the list:
[image]

When I comment out the filter in `getNodes` (in packages/web/src/core/stores/deviceStore/index.ts) all nodes are shown in the list.
```TS
          getNodes: (
            filter?: (node: Protobuf.Mesh.NodeInfo) => boolean,
          ): Protobuf.Mesh.NodeInfo[] => {
            const device = get().devices.get(id);
            if (!device) {
              return [];
            }
            const allNodes = Array.from(device.nodesMap.values()).filter(
              (node) => node.num !== get().devices.get(id)?.hardware.myNodeNum,
            );
/*
            if (filter) {
              return allNodes.filter(filter);
            }
*/
            return allNodes;
          },
```

[image]

The problem is caused by the default metrics filter:
[image]

In case of the above node Degersheim the filter is triggered by RSSI being < -20dBm:
[image]

Second example that is not showing is this one. Here the filter is triggered because the battery voltage is > 5V:
[image]

So I guess the default metric filter settings need some range extension to cover real values.

philon-, 2025-09-09:
> A quick check revealed 29 node in the list that report `voltage -0.0010000000474974513`. The voltage filter is the only one that triggers.
> 
> ```
>   "deviceMetrics": {
>     "$typeName": "meshtastic.DeviceMetrics",
>     "batteryLevel": 101,
>     "voltage": -0.0010000000474974513,
>     "channelUtilization": 13.066667556762695,
>     "airUtilTx": 4.975888729095459,
>     "uptimeSeconds": 7691997
>   },
>   "hopsAway": 4
> ```
> 
> All the device that reporting negative are marked "Plugged in".

Thanks for looking into this again. Can you see what hardware these nodes are running?

The "Plugged in" label is generated from the `batteryLevel` field in the `deviceMetrics` (>100%), and is calculated by firmware and not the client itself.

Mictronics, 2025-09-09:
These are the unique hardware types that are reporting voltage < 0.0 V
```
4 'TBEAM' -0.0010000000474974513
43 'HELTEC_V3' -0.0010000000474974513
5 'HELTEC_V2_0' -0.0010000000474974513
81 'SEEED_XIAO_S3' -0.0010000000474974513
88 'XIAO_NRF52_KIT' -0.0010000000474974513
9 'RAK4631' -0.0010000000474974513
```

Mictronics, 2025-09-09:
The negative voltage is reported in the protobufs. Not a web client issue.

[image]
The log is taken right at this [point](https://github.com/meshtastic/web/blob/6772f066ee83bccde632f24a3d70275088e1428c/packages/core/src/meshDevice.ts#L1162) and is the object returned `fromBinary`.