meshtastic/web #724: [Feature Request]:

KF8CPT, 2025-07-19:
### Prerequisites

- [x] I have searched existing issues to ensure this feature hasn't already been requested
- [x] I have checked the documentation to verify this feature doesn't already exist

### Problem Statement

what I am looking for is a web page similar to the client.meshtastec.org that I can manage all of my nodes (assuming they are in range) and (showing their IP address and or Mac)
also having the ability  to single out 1 or 2 to have different settings...
maybe, at some point... to have scripting available to setup new nodes

### Proposed Solution

make a web page similar to the client.meshtastec.org that I can manage all of my nodes (assuming they are in range) and (showing their IP address and or Mac)
also having the ability  to single out 1 or 2 to have different settings...
 to have scripting available to setup new nodes

### Current Alternatives

doing everything manually

### Importance

Nice to have

### Additional Context

it would be nice to (in map view) have the circles the show the short name to ether have a dark or light ring around them so you can tell at a glance how many are RF and/or MQTT

roberthadow, 2025-10-07:
In the old days (before the web server was no longer installed to the device), each tab in my Chrome browser had the URL in the address bar (e.g. 192.168.1.200, 192.168.1.201, 192.168.1.202, and so on).  Now I have six tabs with the same client.meshtastic.org in the address bar.  Click on the tab and sure, you see the shortName and longName.  

I do a lot of my work with bash scripts using the --host 192.168.1.201 form.  It would be nice to have the device URL visible without digging.

danditomaso, 2025-11-06:
@KF8CPT @roberthadow So we dont have multi-node configuration support yet (we'll get there). But, we do have a new connection manager (you can save your connections, and auto connect to your default connection on HTTP if you choose) its in our test environment and available for users to test out (https://client-test.meshtastic.org). Additionally, we just added a PR to modify the document title to reflect which tab you are connected to. I hope these small changes will improve your user experience. 

[image]