meshtastic/web #744: [Bug]: An unhandled error occurred. <Add details here>change from the 272 fw to stable last fw downgrade.

HB3YDM, 2025-07-28:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

T-Beam S3

### Connection Type

Serial, Bluetooth

### Local or Hosted

http://meshtastic.local

### Firmware Version

2.7.2 dwowngrade to the last stable

### Operating System

with the ios smartwath app

### Browser

firmwre upgradeota button without passsword questions

### Expected Behavior

I clicked on the OTA button at the bottom of the firmware section and it happened. I wondered why something like this could happen without any further questions or password entry.

Fehlerdeta
Error message:
Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.

Stack trace:
NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90550)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90851)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90936)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90438)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90936)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)

### Actual Behavior

I clicked on the OTA button at the bottom of the firmware section and it happened. I wondered why something like this could happen without any further questions or password entry.
Error message:
Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.

Stack trace:
NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90550)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90851)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90936)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90438)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90936)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)

### Steps to Reproduce

go to firmwre
go to OTA
formware upgread without password or an other question ???

### Relevant console output

```Shell
NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90550)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90851)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90936)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90438)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
    at GE (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90936)
    at qr (https://client.meshtastic.org/index-Cif3-n1Z.js:48:90001)
```

### Additional Context

Pleas make an Question befor reset or implement an Second passwort place.

HB3YDM, 2025-07-28:
[image]

HB3YDM, 2025-07-28:
Please make under send Reboot OTA a password question!

danditomaso, 2025-08-09:
@HB3YDM This has been fixed in our latest release today. We now have a prompt prior to taking any action. You should see the changes in our production environment in the next day or two. 

[image]

closed 2025-08-09 by danditomaso