meshtastic/web #782: [Bug]: Unable to add channels in WebUI

ghxstxch, 2025-08-15:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Heltec V3

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

v2.6.11.60ec05e

### Operating System

All

### Browser

All

### Expected Behavior

When adding a new channel and hitting submit, for entered channel settings to save. Can edit existing, but unable to add new. 

### Actual Behavior

Entered channel settings do not save. 

### Steps to Reproduce

Open WebUI via local OR hosted addresses.
Enter in new channel settings under 'Channe'
Hit submit, refresh page, new settings did not save.

### Relevant console output

```Shell

```

### Additional Context

Above answers for devices are a little misleading - happens across all my devices, and on local AND hosted versions of the WebUI. I do not believe connection type matters either.

ghxstxch, 2025-08-15:
> 
> Currently the disconnect handling is not perfect, so sometimes when the node reboots (eg. saving radio config) the connection closes and the client doesn't automatically reconnect - could this be the case?

I don't think this is my issue, but suggesting looking at developer's tools did give me more things to poke and try. Here's what I see in my console:

[image]

Disabling position reporting does nothing, setting it to a different value does nothing either. This was also after power cycling the Meshtastic device and clearing cache/data from the browsers I tested.

philon-, 2025-08-15:
The screenshot indicates that you are not using the latest version of the web client. Please see if you have the same issue using [https://client.meshtastic.org/](https://client.meshtastic.org/) for the latest stable version or [https://client-test.meshtastic.org](https://client-test.meshtastic.org) for the latest unstable version.

ghxstxch, 2025-08-15:
> The screenshot indicates that you are not using the latest version of the web client. Please see if you have the same issue using https://client.meshtastic.org/ for the latest stable version or https://client-test.meshtastic.org for the latest unstable version.

This is off the one bundled with the Home Assistant integration but I did include in my original issue: 

[image]

philon-, 2025-08-15:
Could you provide log output using client or client-test? 

It seems like the Home Assistant plugin bundles 2.5.4 which was released in November 2024.

ghxstxch, 2025-08-15:
> Could you provide log output using client or client-test?
> 
> It seems like the Home Assistant plugin bundles 2.5.4 which was released in November 2024.

I just tested again - I'm not quite sure what I was doing wrong before but you are correct - it does work on both client. and client-test. hosted versions. Like you mentioned, there is no confirmation, and I do have to navigate away from that page for the new channel name to reflect, but it does get saved.

I noticed there was a beta for it updated a few weeks ago, seems that bundles a newer version than the stable and indeed, it works and even has a nice little confirmation toast. 

My apologies for creating an issue that isn't related to the actual project!

closed 2025-08-15 by ghxstxch