meshtastic/web #816: [Bug]: An unhandled error occurred. <Add details here>

14vd165, 2025-08-31:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

T-Lora v1

### Connection Type

Serial

### Local or Hosted

http://meshtastic.local

### Firmware Version

2.7.7

### Operating System

WIN10

### Browser

CHROME

### Expected Behavior

C'est un peu gênant...
Nous sommes vraiment désolés, mais une erreur s'est produite dans le client Web qui a provoqué son plantage. <br /> Cela n'est pas censé se produire et nous travaillons dur pour y remédier.

La meilleure façon d’éviter que cela ne se reproduise, à vous ou à quelqu’un d’autre, est de nous signaler le problème.

Veuillez inclure les informations suivantes dans votre rapport :

Ce que vous faisiez lorsque l'erreur s'est produite
Ce à quoi vous vous attendiez
Que s'est-il réellement passé ?
Toute autre information pertinente
Vous pouvez signaler le problème sur notre [GitHub](https://github.com/meshtastic/web/issues/new?title=%5BBug%5D%3A+An+unhandled+error+occurred.+%3CAdd+details+here%3E&template=bug.yml&logs=NotFoundError%3A+Failed+to+execute+%27removeChild%27+on+%27Node%27%3A+The+node+to+be+removed+is+not+a+child+of+this+node.%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90562%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90863%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90948%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90450%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90948%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29)

Retour au [tableau de bord](https://client.meshtastic.org/)

Erreur de Chirpy the Meshtastic
Détails de l'erreur

### Actual Behavior

BLOCAGE

### Steps to Reproduce

C'est un peu gênant...
Nous sommes vraiment désolés, mais une erreur s'est produite dans le client Web qui a provoqué son plantage. <br /> Cela n'est pas censé se produire et nous travaillons dur pour y remédier.

La meilleure façon d’éviter que cela ne se reproduise, à vous ou à quelqu’un d’autre, est de nous signaler le problème.

Veuillez inclure les informations suivantes dans votre rapport :

Ce que vous faisiez lorsque l'erreur s'est produite
Ce à quoi vous vous attendiez
Que s'est-il réellement passé ?
Toute autre information pertinente
Vous pouvez signaler le problème sur notre [GitHub](https://github.com/meshtastic/web/issues/new?title=%5BBug%5D%3A+An+unhandled+error+occurred.+%3CAdd+details+here%3E&template=bug.yml&logs=NotFoundError%3A+Failed+to+execute+%27removeChild%27+on+%27Node%27%3A+The+node+to+be+removed+is+not+a+child+of+this+node.%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90562%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90863%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90948%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90450%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90948%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29)

Retour au [tableau de bord](https://client.meshtastic.org/)

Erreur de Chirpy the Meshtastic
Détails de l'erreur

### Relevant console output

```Shell
NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90562)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90863)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90948)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90450)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90948)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
```

### Screenshots

C'est un peu gênant...
Nous sommes vraiment désolés, mais une erreur s'est produite dans le client Web qui a provoqué son plantage. <br /> Cela n'est pas censé se produire et nous travaillons dur pour y remédier.

La meilleure façon d’éviter que cela ne se reproduise, à vous ou à quelqu’un d’autre, est de nous signaler le problème.

Veuillez inclure les informations suivantes dans votre rapport :

Ce que vous faisiez lorsque l'erreur s'est produite
Ce à quoi vous vous attendiez
Que s'est-il réellement passé ?
Toute autre information pertinente
Vous pouvez signaler le problème sur notre [GitHub](https://github.com/meshtastic/web/issues/new?title=%5BBug%5D%3A+An+unhandled+error+occurred.+%3CAdd+details+here%3E&template=bug.yml&logs=NotFoundError%3A+Failed+to+execute+%27removeChild%27+on+%27Node%27%3A+The+node+to+be+removed+is+not+a+child+of+this+node.%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90562%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90863%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90948%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90450%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29%0A++++at+nN+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90948%29%0A++++at+go+%28https%3A%2F%2Fclient.meshtastic.org%2Findex-B-cS7hhl.js%3A48%3A90013%29)

Retour au [tableau de bord](https://client.meshtastic.org/)

Erreur de Chirpy the Meshtastic
Détails de l'erreur

### Additional Context

PAS DE POSSIBILITE DE PROGRAMMATION

philon-, 2025-09-10:
Can you please see if you can reproduce the issue using https://client-test.meshtastic.org/ ?

danditomaso, 2025-11-06:
I appreciate that you reported this issue, thank you. Unfortunately, there isn't enough info here to proceed with debugging. Please try  our test environment https://client-test.meshtastic.org and see if your issue is resolved there.

closed 2025-11-06 by danditomaso